smartin015/continuousprint #247: Octoprint 3.10 not supported

Rylan-Meilutis, 2024-04-24:
**Describe the bug**
The plugin is not functional on octoprint version 3.10.0

**To Reproduce**
Steps to reproduce the behavior:
1. Install OctoPrint 3.10.0
2. Install the plugin

**Expected behavior**
The plugin to not experience an error when starting up

**Screenshots**
[image]

[image]

**System info:**
[octoprint-systeminfo-20240424151748.zip](https://github.com/smartin015/continuousprint/files/15096565/octoprint-systeminfo-20240424151748.zip)

**Browser logs:**
[octoprint_browser_log-1713972112384.log](https://github.com/smartin015/continuousprint/files/15096606/octoprint_browser_log-1713972112384.log)

smartin015, 2024-04-27:
Hey, thanks for the report - this looks like a duplicate of #249. You can follow along with the fix there.

closed 2024-04-27 by smartin015

Rylan-Meilutis, 2024-04-27:
Can confirm that the new fix works, than you so much for speedy work :)