smartin015/continuousprint #249: Continuous Print plugin no longer works after upgrading to 1.10.0

gabriel-taubin, 2024-04-25:
**Describe the bug**
A clear and concise description of what the bug is.

I have been using the Continuous Print plugin for quite a long time (years). During that time I installed multiple Octoprint upgrades without any issues. Yesterday I upgraded to Octoprint 1.10.0 and Continuous Print no longer works. In fact, I cannot print anything, even if I select a file directly, not trying to put it in the queue. Parts of the user interface are frozen. Here you have a screenshot of the web interface

[image]

If I disable the Continuous Print plugin in Plugin Manager, and restart the server, everything works fine (but of course Continuous Print is no longer available). That is, I can print individual files. 

I uninstalled the Continuous Print plugin, rebooted the RPi, and then reinstalled it, but I still get the same problem.

Here you have the System Info Bundle that you requested

[octoprint-systeminfo-20240425103453.zip](https://github.com/smartin015/continuousprint/files/15111708/octoprint-systeminfo-20240425103453.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from 1.9.3 to 1.10.0 with Continuous Print previously installed
2. Or
3. Uninstall Continuous Print, restart server or reboot PI, and install Continuous Print
4. Selecting a file from the Files list does not set up the file for printing
5. Print / Pause / Cancel buttons don't do anything
6. Instead of the Pause or Resume labels, the button shows the two labels together as PauseResume, and the button is taller than the Print and Cancel buttons, I guess because the string is longer and it does not fit in the specified size.
7. After disabling the Continuos Print plugin and restarting the server, everything works properly again.

**Expected behavior**
A clear and concise description of what you expected to happen.
- See above

**Screenshots**
If applicable, add screenshots to help explain your problem.
- One attached above

**System info:**
Follow the instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to generate an OctoPrint system bundle, and attach it to this issue.
- attached above

**Browser logs:**
Collect the console logs for your browser and upload them to this issue. See [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file) for instructions on how to do this for different browsers.

- logs seem to be empty; I have never used this feature

**Additional context**
Add any other context about the problem here.

gabriel-taubin, 2024-04-25:
Are the instructions to downgrade in

https://community.octoprint.org/t/how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image/205

still valid ?

smartin015, 2024-04-26:
Hey all - thanks for the report. I've blocked out some time this weekend to figure out what's going on in the new update that broke compatibility.

smartin015, 2024-04-27:
Okay - I think I narrowed it down to a slight change in settings initialization when the page loads. I pushed a new `2.4.1` release - please give it a try and let me know whether that's fixed things on your ends. Thanks!

RettiSeti, 2024-04-27:
Looks to be fixed, thanks!

TimothyGaray, 2024-04-27:
Looks good to me as well.  Thank you!

closed 2024-04-28 by smartin015