smartin015/continuousprint #254: continous print reports a bug no other info

Studio3508, 2024-04-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System info:**
Follow the instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to generate an OctoPrint system bundle, and attach it to this issue.

**Browser logs:**
Collect the console logs for your browser and upload them to this issue. See [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file) for instructions on how to do this for different browsers.

**Additional context**
Add any other context about the problem here.

Rylan-Meilutis, 2024-04-27:
If you are using octoprint 3.10, it is probably the same issue many of us were facing. An issue was already created and it looks like a fix has been implemented. You can find info about that [here](https://github.com/smartin015/continuousprint/issues/249#issuecomment-2081098039). Hopefully this helps you.

smartin015, 2024-04-30:
Hey @Studio3508 - without any details about the bug, I'm afraid I can't help you. As Rylan-Mellutis says, we recently fixed an issue with the latest version of octoprint. Please try updating and see if that fixes things for you.

I'm going to close this bug, but if you continue to see the bug please add some details and I'll get to work on it. Thanks!

closed 2024-04-30 by smartin015

Studio3508, 2024-04-30:
Thank you. Your fix worked and I’m back up and running.