smartin015/continuousprint #263: Continious print in octoprint with OctoPrint-BambuPrinter plugin

igc43, 2025-03-19:
Hello, Through the OctoPrint-BambuPrinter plugin you can connect a bambulab printer to octoprint "A1mini in my case", and print without problems
But if you use the continius print plugin to send a queue with the final gcode configured, it only sends the first print and then stops, it does not continue with the queue.

The octoprint error log shows the following:

025-03-19 15:12:03,465 - octoprint.util.comm - INFO - Changing monitoring state from "Printing from SD" to "Finishing"
2025-03-19 15:12:03,477 - octoprint.printer.standard.job - INFO - Print job done - origin: sdcard, path: cubogc~1.3mf, owner: None
2025-03-19 15:12:03,498 - octoprint.plugins.continuousprint - WARNING - Suppressing sendGcodeScript(afterPrintDone) as driver is in state <bound method Driver._state_start_clearing of <continuousprint.driver.Driver object at 0x63601400>>
2025-03-19 15:12:03,520 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,530 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2025-03-19 15:12:03,665 - octoprint.plugins.continuousprint - INFO - Wrote file ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:03,668 - octoprint.filemanager.analysis - INFO - Starting analysis of local:ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:03,678 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/inaki/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/inaki/.octoprint/uploads/ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:03,700 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: ContinuousPrint/tmp/continuousprint_success.gcode, owner: None, user: None
2025-03-19 15:12:03,722 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2025-03-19 15:12:03,727 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: ContinuousPrint/tmp/continuousprint_success.gcode, owner: None, user: None
2025-03-19 15:12:03,728 - octoprint.plugins.continuousprint - WARNING - Suppressing sendGcodeScript(beforePrintStarted) as driver is in state <bound method Driver._state_start_clearing of <continuousprint.driver.Driver object at 0x63601400>>
2025-03-19 15:12:03,730 - octoprint.plugins.continuousprint - INFO - _state_start_clearing -> _state_clearing
2025-03-19 15:12:03,733 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2025-03-19 15:12:03,743 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2025-03-19 15:12:03,751 - octoprint.filemanager.analysis - INFO - Terminating analysis subprocess for local:ContinuousPrint/tmp/continuousprint_success.gcode...
2025-03-19 15:12:03,807 - octoprint.plugins.bambu_printer.BambuPrinter.serial - WARNING - Not a valid gcode command "End g-code:"
2025-03-19 15:12:03,815 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,820 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,847 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,860 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,868 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {'origin': 'sdcard', 'file': 'c51e9db81f44533716e3af9ffae24760a8ad64f1', 'elapsed': 369}
2025-03-19 15:12:03,893 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,910 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,930 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,937 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,942 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,953 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,965 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:03,988 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,006 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,019 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,043 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,072 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,089 - octoprint.plugins.continuousprint - INFO - _profile_from_path ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:04,108 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,125 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,193 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,208 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,217 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,223 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,256 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,285 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,299 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,339 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,371 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,388 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,419 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,458 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,473 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - 100% speed adjustment command sent successfully
2025-03-19 15:12:04,516 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2025-03-19 15:12:04,518 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,585 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,614 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,620 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,676 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
2025-03-19 15:12:04,698 - octoprint.util.comm - INFO - Finished in 0.977 s.
2025-03-19 15:12:04,734 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2025-03-19 15:12:04,792 - octoprint.filemanager.analysis - INFO - Starting analysis of local:ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:04,810 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/inaki/oprint/bin/python -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/inaki/.octoprint/uploads/ContinuousPrint/tmp/continuousprint_success.gcode
2025-03-19 15:12:04,861 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: ContinuousPrint/tmp/continuousprint_success.gcode, owner: None
2025-03-19 15:12:04,899 - octoprint.plugins.continuousprint - WARNING - Suppressing sendGcodeScript(afterPrintDone) as driver is in state <bound method Driver._state_clearing of <continuousprint.driver.Driver object at 0x63601400>>
2025-03-19 15:12:04,973 - octoprint.plugins.bambu_printer.BambuPrinter - INFO - command sent successfully
[... cut ...]
Follow the instructions [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) to generate an OctoPrint system bundle, and attach it to this issue.

**Browser logs:**
Collect the console logs for your browser and upload them to this issue. See [here](https://support.shortpoint.com/support/solutions/articles/1000222881-save-browser-console-file) for instructions on how to do this for different browsers.

**Additional context**
Add any other context about the problem here.

jneilliii, 2025-03-19:
Happy to help from the [BambuPrinter](https://github.com/jneilliii/OctoPrint-BambuPrinter) plugin side @smartin015, just not familiar with how your plugin works. From my plugin's side the Bambu printer is basically treated as a virtual SD card, and only specific commands will process through the virtual serial comm from OctoPrint directly, ie starting a print, temperature, fan speed control, etc.

smartin015, 2026-08-27:
Hey folks - as something of an A1 mini owner myself, I'm also interested in getting CPQ to work with it. Once I've closed out #267 I intend to take a look at this one.

jacopotediosi, 2026-08-27:
OctoPrint 2.0.0 will allow you to install "Connector" plugins to natively support other printers via connection methods other than serial. Among the connectors already available is [OctoPrint-BambuConnector](https://github.com/OctoPrint/OctoPrint-BambuConnector). It might be worth trying if that works better.